bitaxeorg/ESP-Miner #1115: Add W-iFi SSID to System display page

Nexus9090, 2025-07-03:
As the title say's

Add Wi-Fi SSID into the System display page ideally above the Wi-Fi Status: 

Or maybe make Wi-Fi status read as:-

**Wi-Fi Status:         SSID (Connected!)**

Thanks!!

duckAxe, 2025-07-03:
Like that?

[image]

Nexus9090, 2025-07-03:
Sorry, finger trouble closing the issue by accident

closed 2025-07-03 by Nexus9090

mutatrum, 2025-07-03:
I think it's superfluous, but at least it's complete. The device needs to be connected to the same Wi-Fi as the browser, so it's sort of stating the obvious.

duckAxe, 2025-07-03:
Done https://github.com/bitaxeorg/ESP-Miner/pull/1104/commits/a421a0c9003a80114ad75136b5228064c18394c7

duckAxe, 2025-07-18:
Merged.