bitaxeorg/ESP-Miner #1118: Error - while update esp-miner.bin and www.bin - Unauthorized

diegorodriguezv, 2025-07-03:
**Describe the bug**
Updates cannot be installed; they run up to 100% and then report an error.
When updating both files, I receive the following error message after the update:

[www.bin](http://www.bin/) -> Error - Http failure response for http://bitaxe/api/system/OTAWWW: 401 Unauthorized
esp-miner.bin -> Unauthorized

This is basically a copy-paste of issue #671. I can't tell if this is a regression or if the bug wasn't actually fixed.

**To Reproduce**
Steps to reproduce the behavior:

- access the web interface http://bitaxe
- install previous version:  v2.8.1
- download Latest Release: v2.9.0
- start Updates
- error message appears and update is unsuccessful

**Expected behavior**
The update should succeed. 

**Hardware (please complete the following information):**
 - Bitaxe HW version: Gamma 601
 - Bitaxe HW vendor: bitcoinmerch
 - ESP-Miner FW version: v2.8.1
 - Hash Frequency: 625
 - Voltage: 5

**Additional context**
This happens using the url: `http://bitaxe`
If I use `http://bitaxe.lan` it works okay. I have my main router configured to use lan as the local domain.
I know you can use the ip address as a workaround but for not-technical-savvy people this can be frustrating. Using `http://bitaxe` is convenient for new users.

mutatrum, 2025-07-04:
You have to open AxeOS with the IP address, not the hostname. Unfortunately, this is a limitation of the current version. See #657.

duckAxe, 2025-07-04:
When I first read about this problem on Discord, I thought that I would put a warning on the AxeOS dashboard and tell the user to access AxeOS by IP, not by hostname.

duckAxe, 2025-07-04:
[image]

mutatrum, 2025-07-05:
Moving it out into a separate app won't change the CORS issues, I guess? But I never understood why #657 was closed with 'impossible'.

skot, 2025-07-05:
IIUC we have to enable CORS to support Swarm. To prevent CSRF attacks we have esp-miner block everything except whitelisted local IP addresses.

If we remove Swarm from esp-miner (to a separate app) then we can disable CORS and support accessing AxeOS via hostname again.

Alternatively I wonder if it's possible to whitelist the bitaxe's hostname?

diegorodriguezv, 2025-07-05:
Sorry if I'm not getting the point... But I can do:
```curl -X PATCH  -H "Content-Type: application/json" -d '{"hostname": "bitaxe"}' http://bitaxe69/api/system```
So this feature is to protect from attacks within the bitaxe web interface?

johnny9, 2025-07-06:
> Alternatively I wonder if it's possible to whitelist the bitaxe's hostname?

It should be possible. I believe when we wrote the network access method we just didn't consider host name.