bitaxeorg/ESP-Miner #1154: In v2.10.0b1's Swarm page, some buttons are grayed out as if disabled but still function

Hylinus, 2025-07-19:
**Describe the bug**
In the Swarm page of FW v2.10.0b1, the "Refresh List" and remove device from list buttons are present but grayed out. They still function as expected, but give the appearance of being disabled/unusable. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Swarm" page.
2. Notice that the "Refresh List" and trashcan (remove device from list) buttons are grayed out, as if disabled.
3. Pressing the button performs the expected action.

**Expected behavior**
The button should be bright colored to denote it being enabled.

**Screenshots & Photos**
"Refresh List" and trashcan (remove device from list) buttons in v2.10.0b1 look grayed out/disabled.
[image]
[image]

"Refresh List" and trashcan (remove device from list) buttons in v2.9.0 do not look grayed out/disabled.
[image]
[image]

**Hardware (please complete the following information):**
 - Bitaxe HW version: Gamma 601
 - Bitaxe HW vendor: Altair Tech
 - ESP-Miner FW version: 2.10.0b1

WantClue, 2025-08-21:
@Hylinus have you tested this on the recent beta ?

Hylinus, 2025-08-22:
> [@Hylinus](https://github.com/Hylinus) have you tested this on the recent beta ?

@WantClue, the issue is still present in v.2.10.0b3. The button(s) look grayed out but are clickable.