bitaxeorg/ESP-Miner #1159: Add pool swap

WantClue, 2025-07-23:
Add a button to swap primary and secondary pool for simplicity

hackbuddy1, 2025-07-23:
can i add it for you @WantClue ?

WantClue, 2025-07-24:
> can i add it for you [@WantClue](https://github.com/WantClue) ?

please PR it if you do have some code

BitcoinKid9, 2025-08-10:
Can pool swap be added to the next update? Is there anyone working on this enhancement?
Thank you!

BitcoinKid9, 2025-08-19:
I did figure out that I could simply manipulate/change the port or host on the primary and fallback would become active. However the feature duckaxe proposed is much better and cleaner. :) It would be even better to have several swaps available. Looks like this one is not going into the next update or is it? Thank you!!!

mutatrum, 2025-11-05:
Fixed by #1187

closed 2025-11-05 by mutatrum