bitaxeorg/ESP-Miner #1188: "Power Fault Detected" miner too sensitive, faulting even on proven good PSU

github12101, 2025-08-13:
**Describe the bug**
"Power Fault Detected" on any ATX PSU I've tried

**To Reproduce**
Connect miner to ATX PSU via DC barrel plug
It will never start mining, instead displaying "Power Fault Detected" in the dashboard

**Expected behavior**
Miner works correctly

**Hardware (please complete the following information):**
Device Model:	Gamma (BM1370)
Firmware Version:	v2.9.0
AxeOS Version:	v2.9.0
ESP-IDF Version:	v5.4.1
Board Version:	601

Voltage, Frequency: happens on ALL frequencies and voltages, even lowest possible ones

Miner is most likely hunting for the tiniest spike of low voltage, and once detected, is throws off mining completely, making it impossible to operate on ATX power supply.
Please consider lowering this sensitivity, so I can operate miner with typical ATX PSU.

Tried GameMax 1050W and Dell 360W, both showing around 5.1V on multimeter, both I've been using for years in PCs, they have proven good 5V supply from their 4-pin molex lines.

skot, 2025-08-19:
Yes that's correct. The 5V Bitaxes have a very narrow VIN range, unfortunately.

Make sure your power connectors are very good. Doesn't matter how many watts your PSU if the connector is too high resistance it will fault the Bitaxe.

WantClue, 2025-09-07:
This issue will be closed. The narrowed down 5V VIN range will not be adapted.

skot, 2025-09-10:
> Since you are enforcing usage of chinese OEM power bricks with 5.3V output, instead of allowing high quality ATX PSUs from working with this device, I won't be using them. Already sold all my devices.

Not sure what you mean. Any PSU that can deliver a steady 5.0V (to the voltage regulator -- beware crappy barrel plugs) at the required current will work well.

I'll close this issue for now, seems to be working as expected.