bitaxeorg/ESP-Miner #1220: unlock OTA

WantClue, 2025-09-06:
require button press to unlock OTA

WantClue, 2025-09-08:
A random number generate on the screen could also possible be a solution

WantClue, 2025-09-10:
> Is it a good idea? All users who do not have physical access to their Axes will not be able to update. And imagine if there is an urgent security update available and people cannot update immediately.

yes thats true, but we're thinking about how to improve the security of the OTA functionallity. Imagine you're in an unsecure wifi (which is obviously not a good idea) anybody could load malicious code onto your bitaxe. atm OTA is not protected at all

404-Panda, 2025-09-10:
OTA should use GPG keys to validate downloads or check hashes of files before download.
Would be easy to add checksum checking for OTA downloads sigs can be released and validated before OTA becomes active. 

Any firmware fails checksum should be rejected. 

Also we do need SSL and self signed certs for bitaxe front end should be a way to upload certificated generated by the user to the dash for use. 

Pin code is useless would be easy to bruteforce must be firmware signature checking and maybe lock down the OTA with a user pass where it can only be used when unlocked by the user.

duckaxe, 2025-09-10:
[TLS support](https://github.com/bitaxeorg/ESP-Miner/pull/726)

skot, 2025-09-10:
We should have a NVS flag to disable button-press OTA security. For those that are convinced about their network security.

Button-press OTA security should be enabled by default. And obvs require a button-press to turn it off.

skot, 2025-09-10:
> Is HTTP Basic Auth a possible solution?

I'm not a fan of requiring all Bitaxe users to remember _yet another_ login password. And just having a default password (like most all miners) defeats the entire point.