bitaxeorg/ESP-Miner #1226: Display: Truncated best diff string

duckaxe, 2025-09-10:
The `Best:` string on the display is truncated if it gets too long. The issue occurred after we [added](https://github.com/bitaxeorg/ESP-Miner/discussions/1085) a space after the number.

**Expected**
`Best: 400.08 M/200.54 G`

**Displayed**
`Best: 400.08 M/200.54`

@mutatrum suggested to add a half space (3 pixels wide instead of 6). It's a good idea to wait until his PR https://github.com/bitaxeorg/ESP-Miner/pull/1214 is merged and benefit from his change to add the half space `\u2009` to the font.

**User Image 1**
[image]

**User Image 2**
[image]

mutatrum, 2025-09-14:
Alternative solution would be to do #1202 and have the numbers be integers internally and have a separate formatter in AxeOS and for the display.

mutatrum, 2025-10-23:
Yeah, display is too small to have spaces. Same as with the temperature, looks weird with the huge wide gap between the digits and °C.

Looking at it more, it's a weird layout on the display, with some units before and some after the value.
```
|---------------------| (21 chars max)
|Hash: 512.31Gh/s     |
|Eff.: 22.12J/Th      |
|Best: 839.36M/739.76M|
|Temp: 55.0°C         |
|---------------------|
```