bitaxeorg/ESP-Miner #1242: Memory leak

mutatrum, 2025-09-20:
It's leaking ~40kb in 24 hours.

[image]

mutatrum, 2025-09-20:
This is heap on the ESP32, so esp-miner. It might have been there already, but the new charts makes it easier to show.

mutatrum, 2025-09-20:
It might just be the statistics task collecting 720 data points.

mutatrum, 2025-09-20:
Confirmed, I let it run for 6 hours on the shortest statistics setting (every 30 seconds for 6 hours):

[image]

It might make sense to switch from dynamic allocation to just allocate on startup, if we then run into issues, we can catch the error and still continue running.

WantClue, 2025-09-20:
agreed, a static allocation makes more sense here, this could get confusing if we actually run into a mem leak issue